protoloft/klipper_z_calibration #110: Issue switching build plates

ksummers92, 2023-07-15:
I have set up autoZ against a smooth Wham Bam PEX sheet on my Enderwire and it works perfectly. However, when I switch to a textured sheet, the offset is always around -0.38 which results in a first layer as shown in the image below.

[image]

My understanding is that the autoZ should accommodate for build sheet changes and there is even mention of potentially manually adjusting closer for textured sheets. However, my experience thus far has been that I need to adjust the offset up by as much as 0.4.

Am I missing something?

TitusLabs, 2023-07-17:
This may depend on your switch and the structure of your textured plate. This may depend on whether the switch is only triggered by the mountains, or can also get into the valleys before it triggers. If you need to correct it upwards, then do it with the: SET_GCODE_OFFSET Z_ADJUST=[some value] MOVE=1

ksummers92, 2023-07-17:
Yeah I understand the workaround. I’m just trying to understand how it can be so far off. If I’m having to manually adjust the offset for everything then this auto Z calibration is not providing much benefit…

TitusLabs, 2023-07-19:
No, if the calibration is not consistent, then there is another problem. Check your mechanics, test your klicky and check for a wrong or wrong configured mesh...

Gray-Stone, 2024-06-10:
I actually have similar issue. but worse. For my textured bed, 1 out of say 20 prints will have this super over squash issue. The other times the print needs a tiny extra squash.

TitusLabs, 2024-06-16:
@Gray-Stone is this only an issue with a textured plate or also with a smooth PEI?

Gray-Stone, 2024-06-17:
@TitusLabs This is only an issue with textured plate. smooth PEI works beautifully. 

However, I noticed it might be a problem with my texture plate being wear out near the probing spot. If I intentionally put the plate on with 5mm+ offset, it will now only show as over squashed (needing + z offset), and I could get some good amount of prints without the offset value changing.

I'm not fully warpping around in my head which direction the probing has shifted, be generally, I think the source of the problem is the wear of the plate. When probing at a fresh spot, it's a lot better. 

I'm wondering if its possible to add a little fuzzyness to the probing sequence. Specially move x and y (few mm) for tiny bit between each sample, this way, the average/medium will be for this small area. Also slow down the plate wear.

TitusLabs, 2024-06-17:
Good observation! That's an interesting feature :thinking:  But I don't think there would be that much demand for it! The configuration is already relatively complex. But, you could even solve it yourself using a macro. The CALIBRATE_Z command has a BED_POSITION parameter...

Gray-Stone, 2024-06-17:
The problem is bed-position is a one time thing per-print. What I am hopping for is change the position on every single sample. Which mean within one execution of calibrate_z, have a fuzzy amount. 
Actually KAMP have some thing similar but not the same, KAMP is still change per-print, and done it by changing the overall bed-mesh area, but fuzzing each sample.
If we can find a way to do fuzz between x number of sample for each location, then it could be applied for QGL, mesh, calibrate_z, or any probing move that touch the bed to prevent wearing out the bed.

TitusLabs, 2024-06-20:
Hm, you would lose accuracy by fuzzing the probing position. And you would need more samples. It's an interesting feature. But would that really be useful?

Gray-Stone, 2024-06-20:
Hmmm, I'm already doing 7 points to reduce error. I guess it would just be reducing the plate wear? I can't tell how bad it would be to keep probing at the same point as I haven't run enough print to see the effect yet.

TitusLabs, 2024-06-21:
Probing will not wear out your plate - but always printing in the center will. And waiting to let the print pop off will extend its life.

TitusLabs, 2026-09-03:
Closing this long-running thread, with the reasoning written down here:
https://github.com/protoloft/klipper_z_calibration/wiki/Troubleshooting#the-squish-changes-between-prints-or-between-build-plates

`CALIBRATE_Z` measures the print surface on every run, so a plate swap is
compensated — thickness and surface alike. In practice the result stays
consistent across build plates, coarse textured ones included. A first layer that
changes from print to print is a mechanical inconsistency, and worn or
poor-quality switches are high on that list: a microswitch's trigger point drifts
as it ages, on the calibration endstop as much as on the probe. Play in the probe
mount or dock, a sticking endstop pin, and residue on the nozzle do the same.

If you just want a different squish on a given surface, apply it on top of the
calibrated offset with `SET_GCODE_OFFSET Z_ADJUST=... MOVE=1` rather than
changing `switch_offset`:
https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It#optional-slicer-adjustment

closed 2026-09-03 by TitusLabs