protoloft/klipper_z_calibration #112: Can't get CALIBRATE_Z to work

fryc88, 2023-07-19:
Hi,
After struggling with initial setup, I guess (with help of some people on discord) I've got it set, but when it comes to CALIBRATE_Z, nozzle moves above of z endstop, lowers down and when nozzle triggered endstop, I've got an error in Klipper saying Probe triggered prior to movement, however behaviour of my both z endstop and Klicky seems to be fine during the usage of printer.
[image]
Any ideas what to do with it?

[config-20230703-211348.zip](https://github.com/protoloft/klipper_z_calibration/files/12099765/config-20230703-211348.zip)

TitusLabs, 2023-07-20:
Ok, then please tune your z endstop position. You don't need to fine tune it as the z calibration will take care of it, but roughly it should fit.

fryc88, 2023-07-20:
What do you mean by tuning it?
I've set my z offset against the print surface, so height of z endstop is about right now, or I'm missing something?

TitusLabs, 2023-07-20:
Oh, sorry, forget about it. I had somehow read a different error. It seems that the endstop is already triggered when it tries to probe on it. Is your wiring ok? Is your nozzle too low while moving to the endstop (set safe_z_height to a higher value)?

fryc88, 2023-07-20:
Endstop is triggered when nozzle touches it, not before. Wiring is ok since it's been printing already before and haven't had any other issues.

fryc88, 2023-07-20:
[](https://youtu.be/W1Jb-zCHgcM) 
That's how it looks like.

fryc88, 2023-07-20:
Ok, other person on Voron discord suggested me to change this parameter
`homing_retract_dist: 3`
I originally had it set to 0. After that it started to calibrate offset.
Now need to find out how to tune it, as calculated value crashed nozzle into my bed :)

fryc88, 2023-07-20:
Ok, I knew I didn't make that setting up on my own. Since I have Sensorless homing + z endstop + klicky I knew I saw it somewhere that suggest changing that setting to zero, and it was in official Voron docs regarding sensorless homing :)
[image]
[voron docs](https://docs.vorondesign.com/community/howto/clee/sensorless_xy_homing.html)

IMO it would be worth mentioning that in your configuration/installation to double check that setting, even tho i've noticed not many are using sensorless homing (for no reason because it was easiest part of adjusting it since I built this voron out of any other features that I installed/configured)

TitusLabs, 2023-07-20:
Yes, that is correct for sensorless homing (I don't like it much, because the threshold is always a tradeoff between too much force and not working - and the last one can happen in one of one thousand and can result in a damaged printer). But, be careful. You don't use sensorless homing on Z. On the endstop, you need the retract. By the way, 3mm are long - you can reduce it, if you want..

closed 2023-08-21 by TitusLabs