protoloft/klipper_z_calibration #113: CALIBRATE_Z Move out of range

cosminr86, 2023-07-24:
[z_calibration]
nozzle_xy_position:  230, 351
switch_xy_offsets:   225, 330
switch_offset:        0.5
start_gcode:          attach_probe
end_gcode:            dock_probe
bed_xy_position: 175, 175
probing_first_fast: false

After I run calibrate_z, it docks the probe, probe the nozzle on the z endstop, then rise an error: Move out of range: 455.000 681.000 11.760 [0.000]

TitusLabs, 2023-07-24:
Yes, you configured the absolute switch position as offset from the nozzle's position. Either change this to offsets like: switch_xy_offsets: -5,21 or change it to switch_xy_position

avispa4291, 2023-08-21:
Can you show what this looks like in the config file, I'm having the same issue after updating to bullseye

TitusLabs, 2023-08-21:
As I said, either configure the absolute position of the switch with switch_xy_position or the offsets from the nozzle position with switch_xy_offsets.

And, if you get a "move out of range" error, also check if you enabled the wiggle feature with wrong values. Otherwise your dock/undock macros may be wrong.

avispa4291, 2023-08-21:
I have it configured that way but still get the error
```[z_calibration]
nozzle_xy_position: 230,350
#probe_nozzle_x: 230
#probe_nozzle_y: 350
#   The X and Y coordinates (in mm) for clicking the nozzle on the
#   Z endstop.

switch_xy_offsets: 226,330
#probe_switch_x: 226 # subtract ROUGHLY 3.75 from probe_nozzle_x to start
#probe_switch_y: 330  # subtract ROUGHLY 19.25 from probe_nozzle_y to start
#   The X and Y coordinates (in mm) for clicking the probe's switch
#   on the Z endstop.

#probe_bed_x: default from relative_reference_index of bed_mesh
#probe_bed_y: default from relative_reference_index of bed_mesh
#   The X and Y coordinates (in mm) for probing on the print surface
#   (e.g. the center point) These coordinates will be adapted by the
#   probe's X and Y offsets. The default is the relative_reference_index
#   of the configured bed_mesh. It will raise an error if there is no
#   probe_bed site and no bed_mesh with a relative_reference_index
#   configured.

switch_offset: 0.63	# (was 0.615)lower number = less squish
#   The trigger point offset of the used mag-probe switch.
#   This needs to be fined out manually. More on this later
#   in this section..```

TitusLabs, 2023-08-22:
Please read carefully what I have written twice. The head is at 230,350 and wants to add 226,330 to x and y. The `switch_xy_offsets` takes offset values from your nozzle position. But you defined the absolute values there. Either change `switch_xy_offsets: 226,330` to `switch_xy_offsets: -4,-20` or use `switch_xy_position` like `switch_xy_position: 226,330`.

joydi, 2023-10-14:
> Please read carefully what I have written twice. The head is at 230,350 and wants to add 226,330 to x and y. The `switch_xy_offsets` takes offset values from your nozzle position. But you defined [...]

Hey TitusLabs, that worked very good for me thank you a lot !

closed 2023-10-16 by TitusLabs