protoloft/klipper_z_calibration #114: LDO Voron 2.4 kit's z probe endstop pin sits lower than the bed height

xanderphillips, 2023-07-26:
Should this auto z offset script work when the probe pin height is lower than the bed height?
I can set the location of the probe switch body to be a flat surface on the bottom left or right edge of the probe, but not the back center portion as that would make the extruder be too far over the bed, and drive the nozzle into the bed.

Do I need to just get a probe pin that sits taller than the bed to use this auto z offset configuration, or should I be able to use it this way?

When I try to follow the instructions as-is I end up with an invalid offset value, and that value creeps upwards each time I try to do a calibrate_z.

mikepiko, 2023-07-26:
Can you make a Picture of your Z Endstop Pin ? It should normally stick out above the Bed.

TitusLabs, 2023-07-26:
In principle, it is not a problem if the pin is lower than the bed. This can be adjusted with the endstop position in the configuration.

However, if you can't reach the pin with the probe switch housing, you will have a problem with auto-calibration.

This can be easily solved by using a longer pin, by mounting the entire end stop a little higher or by lowering the probe mount at the print head (if possible). Just to mention some examples..

xanderphillips, 2023-07-27:
Thanks!  I was able to get it finally after tweaking my offset more.

closed 2023-07-27 by TitusLabs