protoloft/klipper_z_calibration #115: Existing command 'CALIBRATE_Z' not found in gcode_macro rename

zsmbrvr, 2023-07-30:
After adding 
```
[gcode_macro CALIBRATE_Z]
rename_existing: BASE_CALIBRATE_Z
gcode:
    {% set bed_position = params.BED_POSITION|default('None') %}
    G28                         # can also be a conditional homing macro
    M117 Z-Calibration..
    ATTACH_PROBE                # a macro for fetching the probe first
    {% if bed_position != 'None' %}
      BASE_CALIBRATE_Z BED_POSITION={bed_position}
    {% else %}
      BASE_CALIBRATE_Z
    {% endif %}
    DETACH_PROBE                # and parking it afterwards (or DOCK_PROBE in klicky macros)
    M117
```

I get the error "Existing command 'CALIBRATE_Z' not found in gcode_macro rename"

Not sure what I am doing wrong.

TitusLabs, 2023-07-30:
That means, that the plugin is not installed. Or, you did not restart the Klipper service after installation/file copy.

zsmbrvr, 2023-07-30:
I did both, what I also did is adding [z_calibration] in a separate file so the module wasn't loading....
Took me too long to find this:
https://www.klipper3d.org/Code_Overview.html#adding-a-host-module
Now that I have [z_calibration] in the printer.cfg file, the command can be found...

closed 2023-07-30 by zsmbrvr

TitusLabs, 2023-07-30:
Ah, sorry! Yes, you also need to configure it and if you have a separate file, you need to include it into the printer.cfg file. Great, you found it - best learning :+1:

zsmbrvr, 2023-07-30:
Strange. I was including it, but it wasn't working.
I moved the config details into the printer.cfg and all is good.