protoloft/klipper_z_calibration #117: Avoid z_calibration if already set

KiwiChrist, 2023-08-12:
Is there a way to avoid setting the z_calibration before every print... like my homing or QGL settings in my print_start.cfg...

This does not home if the printer has already been done and set (obviously z is listed so may need to remove it for z to be excluded if it's not to be reset)...
   {% if "xyz" not in printer.toolhead.homed_axes %}
       G28                                                 
   {% endif %}  

This does not do the QGl if the printer has already done and set....
   {% if printer.quad_gantry_level.applied != True %}
      quad_gantry_level 
   {% endif %}

I do not want the z offset to be reset before every print unless it has to. I would like to keep the last z offset settings starting the next print.

How do I write this up in print_start.cfg?

TitusLabs, 2023-08-21:
There is currently no way to do this. I will add this feature...

KiwiChrist, 2023-08-21:
Thanks Titus. 
Any idea why my z_offest keeps reducing height between print starts? 
Probe accuracy is not an issue. 
Currently I'm restarting the firmware between prints which is successfully eliminating the issue.

I have looked through the cfg's and cannot for the life of me find anywhere that is reduces the z_offest.

TitusLabs, 2023-08-21:
Sorry, I was wrong! You can do like this:
```
{% if printer.z_calibration.last_query != True %}
  CALIBRATE_Z
{% endif %}
```
I will add this to the documentation..

TitusLabs, 2023-08-21:
You calibrate everything, print and rerun calibrate_z and the resulting offset changes without rerunning homing, QGL/Z-Tilt or mesh?

Is the value even wrong? Or is the offset then still correct? The fact is that the value may well change when the printer has become warm.

KiwiChrist, 2023-08-21:
I have G28... so I'll change it to...
{% if "xyz" not in printer.toolhead.homed_axes %}
  G28                                                   
{% endif %}

I already have....
{% if printer.quad_gantry_level.applied != True %}
   quad_gantry_level 
{% endif %}

I will change calibrate_z to this as suggested to...
{% if printer.z_calibration.last_query != True %}
  CALIBRATE_Z
{% endif %}
SET_GCODE_OFFSET Z_ADJUST={params.Z_ADJUST|default(z_offset_dflt)|float} MOVE=1 

I'll let you know how I get on.

KiwiChrist, 2023-08-22:
All working now as expected. There is no longer a z height reduction between prints... that I can tell ATM.

TitusLabs, 2023-08-22:
Glad to hear that! But, I would recommend calibrating z before every print if the printer is closed and can heat up.

closed 2023-08-22 by TitusLabs