protoloft/klipper_z_calibration #119: is it possible to add the z_offset in the purge line?

Gil80, 2023-08-20:
my current macro for purge line is like this:
```
   G90 E0        ; reset extruder distance
    G1 E20 F60    ; purge 20mm
# Purge sequence
    M117 Purge line...
    G1 X1 Y20 Z0.3 F5000.0           ; Move to start position
    G1 E5 F300                       ; Prime the nozzle
    G1 X1 Y200 F1500.0 E15           ; Draw the first line
    G1 X1.5 Y200 F5000.0             ; Move to side a little 
    G1 X1.5 Y30 F1500.0 E30          ; Draw the second line
    G1 Z4 F3000                      ; Move Z Axis up
    M83                              ; Relative E
    G92 E0                           ; Reset extruder distance
    STATUS_PRINTING
```
the problem is that it doesn't take into account the calculated z_offset.
is it possible to add a variable there instead of an absolute value?

TitusLabs, 2023-08-21:
Why do you think it does not take the calculated z-offset into account? If you do the calibration beforehand, it will do it for sure. Or are you using some save/restore gcode state statements in these macros where you should not?

Gil80, 2023-08-21:
Rethinking what I wrote (given I didn't use my Ender3 with klipper z calibration for 4 months :) ) I should have kept the `G1 X1 Y20 Z0` instead of `Z0.3` since it took the nozzle too low on the bed.
Am I right?

TitusLabs, 2023-08-22:
No. If you run the z calibration, every G0/G1 command will take this offset into account. If your printer and your z calibration are set up correctly, a `G1 Z0` should bring your nozzle just right onto your sheet.

For a purge line, I would take a much higher Z value (even higher than 0.3) to be able to extrude a high volume on a short line which does not stick too much to the sheet...

Gil80, 2023-08-22:
but a 0.3 means the nozzle is too close to the bed, no? it should be -0.3 to bring the nozzle a bit higher, isn't it?

TitusLabs, 2023-08-22:
I am not sure I understand you correctly. If you use absolute coordinates (`G90`), the `G1 Z0.3` will move the nozzle 0.3mm above your bed. Provided everything is calibrated correctly.

If you use relative coordinates (`G91`), `G1 Z0.3` will move the nozzle from current position 0.3mm up.

But, this are GCode basics..

Gil80, 2023-08-22:
ok, I'll try to explain what I mean and also what is the confusion.
if I understand correctly (I now doubt that) the z_calibration is all about calculating offsets in order to find the correct height for the nozzle so it's just in the right distance for a good first layer, that becomes my Z=0, the distance at which clibrate_z believes is the right distance of the nozzle from the bed.

What I experienced with the purge line is that the nozzle was scratching the bed, but as soon as the print started (skirt line) it was perfect 1st layer.
This got me to think that maybe I'm doing something wrong in my macro for the purge line. 
In the z_calibrate it states, positive values means closer to bed, negative values higher from bed, so I thought I should modify to purge line to be -0.3.

As I try to explain myself, I realize I'm wrong because telling to printer to move +0.3 it will do just that.

I'm just not sure why the nozzle was scratching the bed on the purge line sequence.

The below is a portion of the `start_print` macro where it shows the purge line sequence:

```
M117 Moving toolhead to bed origin 0x0.

G1 X0 Y0 Z20 F5000

G92 E0 ; reset extruder distance

G1 E20 F60 ; purge 20mm

## Purge sequence

M117 Purge line...

G1 X1 Y20 Z0.3 F5000.0 ; Move to start position

G1 E5 F300 ; Prime the nozzle

G1 X1 Y200 F1500.0 E15 ; Draw the first line

G1 X1.5 Y200 F5000.0 ; Move to side a little

G1 X1.5 Y30 F1500.0 E30 ; Draw the second line

G1 Z4 F3000 ; Move Z Axis up

M83 ; Relative E

G92 E0 ; Reset extruder distance

STATUS_PRINTING
```

TitusLabs, 2023-08-22:
Yes, the z_calibration sets a GCode offset so that Z=0.0 is exactly where the nozzle touches the bed.

You are not using G90 before your moves in your purge sequence and this is bad because you don't know what was set before. You must understand the GCodes G90/G91! Before every move with G0/G1 you have to be sure that you are in the right state. And if you cannot be sure, you must set it. So, before you move to your purge line, you need to do G90...

Gil80, 2023-08-22:
I'm always confused about what should I use or what is more correct to use, G90 or G91.

TitusLabs, 2023-08-22:
Both are correct, it depends on the move you want to make: relative or absolute.

The macro seems to be correct for me. The first G90 should be enough (if no other macro is changing it in between). But, you need to set a higher Z for your start position! With Z0 it will not be able to extrude any plastic..

Gil80, 2023-08-22:
you mean a higher Z for the purge sequence? if yes, then that's where the confusion is. I was set to 0.3, but the nozzle was scratching the bed.

TitusLabs, 2023-08-22:
Now, with Z0.4 in line 65 it must work. Otherwise, your z calibration is not correct and the following print cannot be correct too (if there is nothing else in the Slicer start sequence). 

If the problem still exists, check the calibration manually by homing, z-tilt, z calibraton and moving the nozzle slowly down to Z0. If it barely touches the bed with Z0, then try your START_PRINT without the mesh lines and add a clear mesh command instead...

closed 2024-06-16 by TitusLabs