protoloft/klipper_z_calibration #120: Wiki Changelog describes need to change "safe_z_home" instead of "safe_z_height"

mk-maddin, 2023-08-26:
The [changelog ](https://github.com/protoloft/klipper_z_calibration/wiki/Changelog)for v1.0.0 (2023-04-25) describes:
"_The clearance is deprecated and is replaced by the safe_z_home due to the change in behavior. The safe_z_home is now an absolute z position to move to_."
and 
"_Move from clearance to safe_z_home_"

But as of [Configuration](https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Configure-It#configurations) information it should be "safe_z_height":
```
clearance: DEPRECATED - please use safe_z_height instead!
#   The distance in mm to move up before moving to the next
#   position. The default is two times the z_offset from the probe's
#   configuration.
```

With replacing clearance with safe_z_home klipper throws error:
"_Option 'safe_z_home' is not valid in seciton 'z_calibration'_"
With safe_z_height it works as expected.
Due to this I am assuming this is a typo within the changelog.

TitusLabs, 2023-08-27:
Hi, thanks for the hint! I've changed it :+1:

closed 2023-08-27 by TitusLabs

NateroniPizza, 2024-09-07:
@TitusLabs , should the second instance of "safe_z_home" be also replaced by "safe_z_height?" I just spent quite a while troubleshooting why it wouldn't work, without realizing there was a difference between the two:
[image]