protoloft/klipper_z_calibration #121: Probe trigger height being used as Zero position

TmRxJD, 2023-09-19:
I have been using this plugin for a while now with no issues, and have been exceedingly happy with it. Last night I made a change to my toolhead and the probe offset is slightly smaller now. Even though I adjusted this to the correct value, every time I go through the calibration process now the nozzle is always about 2mm too high from the bed, but this is the exact height the probe should trigger at. I have no idea what's going on, clearly I've set something wrong but after hours of messing with this I can't figure it out. Any help would be appreciated. My understanding was you should be able to change the bed and nozzle and this plugin will compensate automatically..

TitusLabs, 2023-10-16:
This is very strange. I don't know what you changed at your tool head, but I would have assumed that the calibration should work without any other changes (except for X, Y adjustments at the endstop). Have you found a solution yet? Otherwise, I would go through all the settings and calibration steps again piece by piece and make sure that no mesh is loaded...

closed 2024-06-16 by TitusLabs

TmRxJD, 2024-06-17:
Thank you for the reply. I have somewhat found a solution, manually raising the offset after printing the first layer. I think there is some backlash in the system, and even though I tried antiback lash nuts the problem persists. I've also recently developed a band at a consistent height on anything I try to print, I've never had banding issues before and always bragged about how perfect my prints looked. Here is my start sequence, maybe I'm missing something. I made sure to clear the mesh before starting offset calibration.

[gcode_macro START_PRINT]
gcode:
   {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
   {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(200)|float %}
    #Start bed heating (but don't wait for it)
    SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET={BED_TEMP}
    # Restore max speed/accel/accel_to_decel to their configured values
    SET_VELOCITY_LIMIT VELOCITY={printer.configfile.settings.printer.max_velocity} ACCEL={printer.configfile.settings.printer.max_accel} ACCEL_TO_DECEL={printer.configfile.settings.printer.max_accel_to_decel}
    G21 ; Millimeter units
    G90 ; Absolute XYZ
    M83 ; Relative E
    G92 E0 ; Reset extruder distance
    {% if not 'xyz' in printer.toolhead.homed_axes %}
        G28
    {% endif %}
    Attach_Probe_Lock
    TEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={BED_TEMP}; Wait for bed to reach target temp
    SET_HEATER_TEMPERATURE HEATER=extruder TARGET=100
    BED_MESH_CLEAR
    Z_TILT_ADJUST
    CALIBRATE_Z
    SET_HEATER_TEMPERATURE HEATER=extruder TARGET=175
    BED_MESH_CALIBRATE
    SET_HEATER_TEMPERATURE HEATER=extruder TARGET={EXTRUDER_TEMP}
    Dock_Probe_Unlock
    #Wait for nozzle to reach target temp
    TEMPERATURE_WAIT SENSOR=extruder MINIMUM={EXTRUDER_TEMP}
    {% if BED_TEMP >= 100 %}
        TEMPERATURE_WAIT SENSOR='z_thermal_adjust' MINIMUM={30}
    {% endif %}
    KOMB
    LINE_PURGE

 I've also made sure every single screw I can find is tight. This has always worked great for me, even after the upgrades, then just suddenly started acting up. I have to be missing something. If there's any other information that would be helpful I am happy to provide. Thank you for you time in working with me on the issue.