protoloft/klipper_z_calibration #122: [Feature suggestion] Nozzle wipe on Endstop

todstelzer, 2023-10-02:
I have a suggestion for the macro:

`[gcode_macro CALIBRATE_Z]
rename_existing: BASE_CALIBRATE_Z
gcode:
    M117 Z-Calibration..
    G28 Z
    CLEAN_NOZZLE ;nozzle clean script
    G28 Z
    G1 Z XX f600 ;Your Z-endstop height +0,1mm
    G91 
    {% for _ in range(6) %}
    G1 X+1 f60
    G1 X-1
    {% endfor %}
    G90
    G28 Z
    G1 Z5 f6000
    BASE_CALIBRATE_ Z
    M117 
`

With this code the nozzle gets wiped on the Z_Endstop multiple times to get rid of the last bit of filament.
Not sure you can call the z_endstop height automaticly.

With this the i got 0.02-0.05mm better accuracy.

todstelzer, 2023-10-11:
I had some time with it and mod it to set hotend temp using tap macro.
[gcode_macro CALIBRATE_Z]
rename_existing: BASE_CALIBRATE_Z
gcode:
    {% set TARGET_TEMP = printer.extruder.target %}
    M117 Z-Calibration..
    SET_VELOCITY_LIMIT ACCEL=5000
    SET_VELOCITY_LIMIT VELOCITY=500
    NOZZLE_TEMP_GUARD
    G28 Z
    CLEAN_NOZZLE
    SET_HEATER_TEMPERATURE HEATER=extruder TARGET=140 ;set hotend to 140c
    G28 Z
    M106 S255 ;Part Fan 100%
    G1 z1.1 f600
    G91
    {% for _ in range(5) %}
    G1 X+1.5 f60
    G1 X-1.5
    {% endfor %}
    G90
    G28 Z
    BASE_CALIBRATE_Z
    M106 S0 ;Part Fan
    SET_HEATER_TEMPERATURE HEATER=extruder TARGET={TARGET_TEMP} ;set previose hotend temp
    G1 Z10 F24000
    G1 X50 Y356
    TEMPERATURE_WAIT SENSOR="temperature_sensor hotend" MINIMUM={(TARGET_TEMP - 5)}
    M117

TitusLabs, 2023-10-16:
This is interesting. Glad to see it's working for you. A single movement with the wiggle function is not enough for that? 

Common practice would be to use a sex bolt endstop with a spring to build up a little more pressure against the nozzle to squeeze the plastic.

By the way, you can get the endstop position with this statement: `printer.configfile.settings['stepper_z'].position_endstop`

todstelzer, 2023-10-24:
Yeah you can use 0.1mm there. You don´t need the endstop height after all because you are doing a g28 z anyway ^^;
There are more moves because the nozzle needs time to cool down. Thats why you can wipe longer.
I suggest to use a ceramic heater to heat the nozzle faster to print temp afterwards.

closed 2024-06-16 by TitusLabs