protoloft/klipper_z_calibration #123: Do I need to move xy to a safe position myself after G28?

rushairer, 2023-10-11:
```yaml

# This macro was provided by discord user Garrettwp to whom i give my thanks for sharing it with me.
# I have tweaked it a lot.
# They are based on the great Annex magprobe dockable probe macros "#Originally developed by Mental,
# modified for better use on K-series printers by RyanG and Trails", kudos to them.
# That macro as since evolved into a klipper plugin that currently is pending inclusion in klipper,
# more information here, https://github.com/Annex-Engineering/Quickdraw_Probe/tree/main/Klipper_Macros
# User richardjm revised the macro variables and added some functions, thanks a lot
# by standing on the shoulders of giants, lets see if we can see further
#
# the current home for this version is https://github.com/jlas1/Klicky-Probe
# the 1000 values below is to give an error instead of doing something wrong, hopefully, this won't be used is a printer larger than 1 meter

[gcode_macro _User_Variables]
variable_verbose:             True    # Enable verbose output
variable_debug:              False    # Enable Debug output
variable_travel_speed:         200    # how fast all other travel moves will be performed when running these macros
variable_move_accel:          1000    # how fast should the toolhead accelerate when moving
variable_dock_speed:            50    # how fast should the toolhead move when docking the probe for the final movement
variable_release_speed:         75    # how fast should the toolhead move to release the hold of the magnets after docking
variable_z_drop_speed:          10    # how fast the z will lower when moving to the z location to clear the probe

variable_safe_z:         	    10    # Minimum Z for attach/dock and homing functions
# if true it will move the bed away from the nozzle when Z is not homed
variable_enable_z_hop:        True    # set this to false for beds that fall significantly under gravity (almost to Z max)

variable_max_bed_y:            310    # maximum Bed size avoids doing a probe_accuracy outside the bed
variable_max_bed_x:            310    # maximum Bed size avoids doing a probe_accuracy outside the bed

# if a separate Z endstop switch is in
# use, specify the coordinates of the switch here (Voron).
# Set to 0 to have the probe move to center of bed
variable_z_endstop_x:         254
variable_z_endstop_y:         313

#Check the printer specific documentation on klipper Dock/Undock configuration, these are dummy values
#dock location
variable_docklocation_x:      178    # X Dock position
variable_docklocation_y:      303    # Y Dock position
variable_docklocation_z:      18.96    # Z dock position (-128 for a gantry/frame mount)

#The following variables are used if the dock is deployed and retracted via a servo motor
variable_enable_dock_servo:  True    # Set to true if your klicky dock is servo-controlled
variable_servo_name:         'probe_servo'    # The name of the dock servo defined in printer.cfg under [servo]
variable_servo_deploy:          0    # This EXAMPLE is the value used to deploy the servo fully
variable_servo_retract:         90    # This EXAMPLE is the value used to retract the servo fully (initial_angle in [servo] config)
variable_servo_delay:          250    # This is a delay to wait the servo to reach the requested position, be carefull with high values

#Dock move, final toolhead movement to release the probe on the dock
#it's a relative move
Variable_dockmove_x:             40
Variable_dockmove_y:             10
Variable_dockmove_z:             20

#Attach move. final toolhead movement to attach the probe on the mount
#it's a relative move
Variable_attachmove_x:           0
Variable_attachmove_y:          40
Variable_attachmove_z:           0

#Umbilical to help untangle the umbilical in difficult situations
variable_umbilical:          False    # should we untangle the umbilical
variable_umbilical_x:           15    # X umbilical position
variable_umbilical_y:           15    # Y umbilical position

# location to park the toolhead
variable_park_toolhead:      False    # Enable toolhead parking
variable_parkposition_x:       125
variable_parkposition_y:       125
[... 158 lines cut ...]

After I ran the G28 command, the nozzle stayed at the z endstop switch xy position.
Is not safe before print.

so i add this gcode in [homing_override] of klicky
``` yaml

[homing_override]
...
        _Home_Z_

        #!!!!!!!!!!!!!!!!!!! insert this code after _Home_Z_
        # move xy to safe position after homing
        {% if after_homing_safe_x != 0 %}
           G0 X{after_homing_safe_x} F{travel_feedrate}
        {% endif %}
        {% if after_homing_safe_y != 0 %}
           G0 Y{after_homing_safe_y} F{travel_feedrate}
        {% endif %}

```

Am I doing duplicate work? Is there any other place where similar functionality can be configured?

Thanks！

TitusLabs, 2023-10-16:
With safe_z_home you have only the option of the homing retract setting for Z direction. If you need a different position in X and Y, you can use the option move_to_previous=true - but, this depends on your previous position and might be unpredictable. So, the only and correct place to do it is the homing_override :+1:  Only there you can make sure that it is always called afterwards.

rushairer, 2023-10-18:
thanks @TitusLabs

closed 2023-10-22 by TitusLabs