protoloft/klipper_z_calibration #125: Move out of range during Calibrate_Z

panas78, 2023-10-28:
After updating to latest release i am having error during Calibrate_z

[klippy (2).log](https://github.com/protoloft/klipper_z_calibration/files/13196154/klippy.2.log)

panas78, 2023-10-29:
i also tried 
CALIBRATE_Z BED_POSITION={75.00, 187.50}
but again it tries to move out side of bed dimensions

panas78, 2023-10-30:
also tried to set 
bed_xy_position: 175.00, 187.50
in [z_calibration]
with no luck

new klipper.log uploaded
[klippy (5).log](https://github.com/protoloft/klipper_z_calibration/files/13207311/klippy.5.log)

TitusLabs, 2023-10-30:
Hi! The problem is, that you defined the switch_xy_offsets as absolute values. Either, you change it to switcch_xy_offsets=-3.75,-19.25 (if your values are correct) or by using the right property for that: switch_xy_position=227.25,340.75

closed 2023-10-31 by panas78