protoloft/klipper_z_calibration #127: Z Calibration Off, Printing Above plate

PresleyHudson, 2023-11-13:
[image]

I recently got Klicky and an endstop pin setup to get z calibration set up. I spent the day working though it to get everything I needed, I went to start a print and it is printing way above my build plate it thinks its at 0.2 but it is not and when only homing it is correct.

when manually measuring the z offset it was closer to the Probe value, or the formula alone without subtracting from Probe.
one thing I was wondering was where the Probe value was pulled from?

the position of my z endstop is 4.1 and like i had said before when i home and move it to z 0.2 its measured correct with feeler gauge

Asiak3, 2023-11-18:
Im having the same issue, i think the solution is to have the start gcode like this:
    SET_GCODE_OFFSET Z=0  # I dont know if this is needed
    BED_MESH_CLEAR             # I dont know if this is needed

    M117 Homing
    G90 # use Absolute Positioning 
    SMARTHOME # Home neded axes, Z allways
    
    Attach_Probe
    Z_TILT_ADJUST
    Dock_Probe

    CLEAN_NOZZLE #run clean nozzle macro
    
    Attach_Probe
    
    CALIBRATE_Z
    
    Dock_Probe   # before i´ve done the mesh directly without docking the probe and now docking all works fine for me
    
    BED_MESH_CALIBRATE    
    BED_MESH_PROFILE Load=default
    
    #G1 X249 Y45 F6000 ;nozzle clean start position
    
    #M109 S{T_EXTRUDER} # wait till Hotend temp is reached, then continue
        
    M117 Purge
    
    LINE_PURGE

For me the solution was to Dock the probe after  CALIBRATE_Z and pickup again for BED_MESH_CALIBRATE.

TitusLabs, 2023-11-18:
Did you test the offset before starting a print? Was that correct? Otherwise, I would check the PRINT_START macro as @Asiak3 suggested. The reset of the offset is not needed, as the plugin does this anyway. But clearing any mesh, calculating the z offset and probing a new mesh (if needed).

PresleyHudson, 2023-11-19:
If I understand correctly, yes I believe I tested the offset before starting a print using a 0.2mm feeler gauge and I believe it was fine.

in the mean time I have just been running with the offset that was calibrated not running it every print, it has been great.
once I get the revo and im swapping nozzles I will look into this again.

Asiak3, 2023-11-19:
i also forgot to mention that i since i added to the bed_mesh section the 
zero_reference_position:
#   An optional X,Y coordinate that specifies the location on the bed
#   where Z = 0.  When this option is specified the mesh will be offset
#   so that zero Z adjustment occurs at this location.  The default is
#   no zero reference.

it seams to have improved. i dont know if it was the store probe action or the zero_reference_position

Asiak3, 2023-11-19:
[bed_mesh]
speed: 150
horizontal_move_z: 6
mesh_min: 10, 10 #45, 10 #25, 10
mesh_max: 220, 199 #205, 217
probe_count: 7, 7
mesh_pps: 2,2
algorithm: bicubic
fade_start: 0.8
fade_end: 10
fade_target: 0
zero_reference_position: 117, 117

Rhodani, 2023-11-19:
I have the same problem. I am using KAMP and a klicky probe. I just swapped to another buildplate and now it is 0.4 to high. If I do a z_calibrate it is spot on. So there has to be something wrong in my config. Has anyone found the problem?

TitusLabs, 2023-11-21:
The PRINT_START macro seems to be ok.

It's important to define the zero_reference_position in your mesh configuration (e.g. the middle of the bed).

But, if you use KAMP or any other adaptive bed mesh, then you need to call the CALIBRATE_Z with the BED_POSITION parameter with the current zero_reference_position used by the adaptive mesh.

Rhodani, 2023-11-21:
Could you please give me an example? I am using KAMP with adaptive bed
mesh. I have no idea how to do what you wrote, but it seems like it is
something that is fast to implement.

TitusLabs, 2023-11-21:
I have never dealt with KAMP or adaptive mesh. I even don't use mesh at all. But, this should be documented in KAMP. You will see that it works correctly if the z calibration is always done in the center of the current (smaller) mesh.

PresleyHudson, 2023-11-22:
I was not aware of this zero_reference_position option I needed to set for use of probe and endstop for z.
I will get this added and test again.

TitusLabs, 2023-12-03:
I'm glad it's working now. The calibration adds a gcode offset. It's nothing permanent.

PresleyHudson, 2024-03-08:
Sorry to bring this back up again, been having trouble getting it to be consistent it seems to keep raising up on me even when I make adjustments I believe should lower it.

I did miss the part about using BED_POSITION, should the CALIBRATE_Z be ran after making the bed mesh with bed_position?

I have thought I have followed all the tutorials I've found but not getting the same results...

PresleyHudson, 2024-03-08:
I think I've got it working more consistently now,

I'm running my Bed Mesh (KAMP) before the Calibrate Z and have added the BED_POSITION Param with the same cords as the zero reference point like you had said before (and in the documentation) but I had missed

closed 2024-06-16 by TitusLabs