protoloft/klipper_z_calibration #128: klippy/extras/z_calibration.py untracked

NikosSprocket, 2023-11-13:
I was away for a few months so went I just updated my Voron 2.4 from Mainsail 2.6.1 to 2.8.0 I found klipper v0.12.0-0 is complaining that "klippy/extras/z_calibration.py" is now an untracked file.

Just to make sure I run git pull (forward) on the "klipper_z_calibration" repo and re-run install.sh; just in case...

At the end added "klippy/extras/z_calibration.py" in ~/klipper/.git/info/exclude.

Am I doing something wrong or the new version of Mainsail/Moonraker are more ... chatty and these untracked files where always there?

PS. I love your plugin and I am **not** moving to TAP!

TitusLabs, 2023-11-18:
Haha, thank you very much. I'm glad you like it. You haven't done anything wrong. It has always been an untracked file because it does not exist in the Klipper repository. However, a new file does not prevent Klipper updates from being loaded. It is just a new "feature" of Mainsail to show this now... and excluding it by this way is a clean way :+1:

roykrikke, 2023-12-20:
> Haha, thank you very much. I'm glad you like it. You haven't done anything wrong. It has always been an untracked file because it does not exist in the Klipper repository. However, a new file does [...]

So no why to get this fixed somehow?
[image]

closed 2023-12-20 by NikosSprocket

NikosSprocket, 2023-12-20:
As explained by @TitusLabs there is nothing to fix.

If you want to stop getting these notifications then add the following line:

klippy/extras/z_calibration.py

in this file "~/klipper/.git/info/exclude".