protoloft/klipper_z_calibration #130: Code issue in clearance variable

Eremius, 2023-12-18:
I am receiving an issue from Klipper
`Unable to parse option 'clearance' in section 'z_calibration'`
In the autoz.cfg file the line is 
`clearance: 2 * z_offset from the "probe:z_offset" section`

I am using a euclid probe.  I am not sure if this matters but he euclid probe include (which contains the z_offset value) precedes the autoz.cfg include.

I did just notice that all the text after clearance is lt blue instead of the orange
[image]

I suspect this is a python spacing issue but I am not sure what it should be.

Eremius, 2023-12-18:
...and the log seems to show it's a casting error

```
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 35, in _get_wrapper
    v = parser(self.section, option)
  File "/usr/lib/python3.7/configparser.py", line 823, in getfloat
    fallback=fallback, **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 808, in _get_conv
    **kwargs)
  File "/usr/lib/python3.7/configparser.py", line 802, in _get
    return conv(self.get(section, option, **kwargs))
ValueError: could not convert string to float: '2 * z_offset from the "probe:z_offset" section'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 130, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/z_calibration.py", line 469, in load_config
    return ZCalibrationHelper(config)
  File "/home/pi/klipper/klippy/extras/z_calibration.py", line 27, in __init__
    self.clearance = config.getfloat('clearance', None, above=0.)
  File "/home/pi/klipper/klippy/configfile.py", line 67, in getfloat
    note_valid=note_valid)
  File "/home/pi/klipper/klippy/configfile.py", line 40, in _get_wrapper
    % (option, self.section))
configparser.Error: Unable to parse option 'clearance' in section 'z_calibration'
webhooks client 4134713488: New connection
webhooks client 4134713488: Client info {'program': 'Moonraker', 'version': 'v0.8.0-209-g4235789'}
```

TitusLabs, 2023-12-18:
Hi, this is the default from the documentation. You need to define a (float) number if you want a different value. And please use the safe_z_height property since clearance is already deprecated.

Eremius, 2023-12-18:
Should this value be picked up automatically by default or do you need to explicitly set it?

closed 2024-06-16 by TitusLabs