protoloft/klipper_z_calibration #133: Is it possible to have a dedicated Z Switch pin instead of just using the Probe Pin?

wil-sys, 2024-01-06:
I have Klipper on an Anycubic Kobra 2 Neo, and its stock Auto Z-Offset has a little switch on the end of the bed. The stock firmware first probes the middle of the bed, and then goes over to the dedicated switch to get the Z offset. Is there a way to do that with this plugin?

AJolly, 2024-02-02:
We'd have to figure out the pins.  The virtual z-endstop is giving me all sorts of issues on this machine.  It's extremely unreliable:  
[image]

wil-sys, 2024-02-04:
That’s a lot of Deviation, I noticed that a lot on the stock firmware Z offset display but not much with klipper. As for the pin I believe it’s PA8, but I don’t see any way to do it regardless with this plugin.

AJolly, 2024-02-04:
Button is PA10 figured that out today. haven't figured out which ones for the filament run out sensor.

wil-sys, 2024-02-04:
Now that we know the switch pin is PA10, is there a way to get the auto Z offset function working using both the probe and button?

AJolly, 2024-02-04:
If I'm understanding the requirements right for this project, we need:

1) A z endstop where the tip of the nozzle drives on a switch (like the standard Voron V1/V2 enstop). It does not work with the virtual pin of the probe as an endstop!  - We now know the pin pin PA10, nozzle dives and pushes the button to activate the switch.

I need to figure out the rest of the config to get klipper to properly use this as the z-endstop.  Right now the klipper configs are using the same as marlin - using the induction sensor.  

Poking around the marlin configs, it looks like the correct position to test at is - 
#define NOZZLE_X    60 		      //calibration module x pos
#define NOZZLE_Y    240			  //calibration module y pos

2) A magnetic switch based probe at the print head - instead of the standard inductive probe (e.g. [this one from Annex](https://github.com/Annex-Engineering/Quickdraw_Probe), or the popular drop in replacement [KlickyProbe](https://github.com/jlas1/Klicky-Probe) with many mounting options)
This would need to be added to the kobra2 neo, since right now it uses a crappy induction sensor.  
I need to go learn this and figure otu what we'd need to do.

TLDR: It's possible to get both auto z-offset and this project working, just have to spend some time figuring it all out.

AJolly, 2024-02-04:
This works for me for using the button as the endstop.  

[stepper_z]
step_pin: PB0
dir_pin: !PB1
enable_pin: !PA15
microsteps: 16
rotation_distance: 8
position_min: 0
position_max: 220
homing_speed: 2
second_homing_speed: 1
endstop_pin: ^PA0

#zhome probing. Probably dont change this
[safe_z_home]
home_xy_position: 56,238.5
speed: 250
z_hop: 10
z_hop_speed: 15

Neuroplant, 2024-06-06:
Hi,
So am I right that it is not possible to make this work without additional hardware?
Don't get me wrong, but if your printer is equipped with a Z-probe, it does not have a (redundant) Z-endstop.
If you add a Z-probe yourself, you usually reuse the pin of the (then no longer needed) Z-endstop.
Does anyone use this add-on? Do you always need three pins? Or how is this solved on other printers?

Neuroplant

TitusLabs, 2024-06-16:
I don't know anything about your printer. But what @AJolly  has written seems plausible :+1: 

You need a z-enstop pin for the nozzle and the magnetic switch probe (not used as virtual z endstop).

TitusLabs, 2024-06-16:
@D3ltorohd you need to set a position_endstop value in your stepper_z section :+1:

amg-web, 2025-07-27:
@AJolly 
> This works for me for using the button as the endstop.

which printer is it?

amg-web, 2025-08-04:
> This works for me for using the button as the endstop.

confirm, it works in simple way.
without probing with a magnet probe and other operations before starting as described in the documentation.

TitusLabs, 2026-09-03:
2.0.0 adds exactly this: an optional `endstop_pin` in `[z_calibration]` for a Z
switch that isn't the endstop of any stepper. It's meant for setups that home Z
through `probe:z_virtual_endstop` — @printicus, that covers your toolchanger
case: keep homing via the probe and point `endstop_pin` at the Sexbolt.

With `endstop_pin` set you must configure `nozzle_xy_position`; `[safe_z_home]`
is no longer used as a fallback, because it positions the probe over the bed,
not the nozzle over the switch.

printicus, 2026-09-04:
> 2.0.0 adds exactly this: an optional `endstop_pin` in `[z_calibration]` for a Z switch that isn't the endstop of any stepper. It's meant for setups that home Z through `probe:z_virtual_endstop` — [...]

He he... my post was a long time ago, before I found you on discord, Thank you. The new version works great.

TitusLabs, 2026-09-04:
Yeah, sorry. I don't have much time for this. You just caught me at a good moment. I'm glad I could help you!