protoloft/klipper_z_calibration #134: Automatic detection of the offset between nozzles of different tools.

Octopusmode, 2024-01-07:
I've read the source code, but I haven't figured out if this plugin can be used to automatically detect the Z difference between the nozzles of each extruder.

[image]

I have the print bed itself as a Z0 sample on my printer. The print head has a switching head installed as on Stratasys. I.e. to determine the Z offset between the nozzles, you just need to poke them alternately on the table.

Is it possible to use this plugin to solve this problem?

TitusLabs, 2024-06-16:
Hi, that is not possible at the moment. But, maybe we need that support? There are also more and more IDEX out there :thinking:

TitusLabs, 2026-09-03:
Measuring per-tool nozzle offsets is out of scope; this plugin calibrates one
nozzle against the bed.

2.0.0 takes the other side of the problem instead: it warns when `CALIBRATE_Z`
runs while a tool with active offsets is selected, and `offset_gcode` lets you
hand the result to your toolchanger instead of writing `GCODE_OFFSET`. Calibrate
with the reference tool, apply your measured tool offsets on top. Closing.

closed 2026-09-03 by TitusLabs