protoloft/klipper_z_calibration #136: [FR] Support for IDM Scanner

Travis90x, 2024-02-25:
After Homing and Z_Tilt, I do CALIBRATE_Z and got
No trigger on probe after full movement when the
The toolhead stop a Z=0.5

I use 2 endstop Z (optical) for 2 independent Z axis
and I would use IDM Scanner to autocalibrate Z_Offset.
It is like Beacon sensor, so I no need to Lift head.

Maybe IDM scanner has not value as Probe Triggered or Open, 
but it works if used as an endstop Z (with probe:z_virtual_endstop)
IDM want no [probe] section declared.

[z_calibration]
nozzle_xy_position: 155, 155
switch_xy_offsets:  0, 25
bed_xy_position:   130,130
switch_offset:  2
lift_speed: 5
samples: 3
samples_tolerance: 0.1
samples_tolerance_retries: 3
samples_result: average
safe_z_height: 10
position_min: 0.5
speed: 100
probing_speed: 20
probing_second_speed: 5
probing_retract_dist: 1
probing_first_fast: false

[idm]
serial: /dev/serial/by-id/usb-IDM_614e_2C0013001743565537353020-if00
speed: 5
lift_speed: 5
backlash_comp: 0.5
x_offset: 0
y_offset: 25
trigger_distance: 2
trigger_dive_threshold: 1.5
trigger_hysteresis: 0.006
cal_nozzle_z: 0.1
cal_floor: 0.1
cal_ceil: 5
cal_speed: 1
default_model_name: default
mesh_main_direction: x
mesh_cluster_size: 3
mesh_runs: 2

TitusLabs, 2024-06-16:
The auto calibration would not work with the z-endstop at the axis - or how do you use the endstops? And when you use a Beacon like sensor, why do you want to use it at all?

TitusLabs, 2024-06-17:
Hmm, I don't see what I could do about it. None of this has anything to do with my auto z calibration :thinking:

Travis90x, 2024-06-17:
[image]

https://youtu.be/oQYHFecsTto
See the video of the wiki, autocalibration_z perform the homing Z with the nozzle on a fixed switch outside the bed.

Then perform 2 measurements with the "probe" (in that case  it use euclid probe, a simple omron microswitch, like the klicky).

I can't do these 2 measurements with IDM scanner because it is not defined as a Probe for klipper (it uses 3rd part idm.py)

With those measurements we can autocalibrate the z_offset (the real/new distance of the nozzle from the bed).

(In my case I need only the measurements of the z_endstop and the IDM, so calculate the differece and set it as z_offset.
Otherwise I have to find a way to perform G28 Z with the endstop PIN, and then a G28 Z with the virtual:endstop )

https://gitee.com/NBTP/IDM/tree/master

TitusLabs, 2024-06-17:
Hm, ok. That would be enough in theory. However, I come from the Voron world with a closed chamber with temperatures of 60 degrees and more (probes are even hotter). There is no probe that can cope with this with sufficient tolerance - you can only achieve this with a simple switch. Electrical properties change too much. Beacon can't do that either. Even worse, you can't even measure the PEI surface with Beacon. This is not automatic calibration to my understanding.

Travis90x, 2024-06-17:
Sure, my case is very particular. Probably, to give you an example closer to your case, what I need would be similar to someone who has a Voron and uses the new BTT Eddy in the "[Coil version](https://cdn.shopify.com/s/files/1/1619/4791/files/Eddy_Series_product_page20240429_23.png?v=1714359282)" (which cannot be used as a "virtual:endstop Z", instead of USB version). 
Therefore, I imagine that the user who wants to automatically calibrate the z_offset must perform a Z homing with a switch (not precise) and then find a way to do a probe on the bed with the Eddy (which is very similar to the beacon/IDM) to calibrato the z offset.

Let's leave aside the fact that sensors like the beacon/IDM are not precise with the PEI coating. I know how you feel about this and you are not wrong.

However, just to clarify, I have tried the IDM and the Beacon by doing a mesh with the bed at 120°C in a closed chamber preheated for 1 hour and I didn't have any problems (I have the VzPrinthead in aluminum and only few parts in Pa12-CF printed in SLS). 
The only advice the IDM manufacturer gives is to turn off the AC bed if >500w (due to AC interference) during the mesh (few seconds).

TitusLabs, 2026-09-03:
Beacon-style scanning probes are out of scope, and 2.0.0 states that
explicitly. They can't give a repeatable absolute distance across the
temperature range this plugin has to work in, which is the whole basis of the
calculation. Closing — the thread already has the full reasoning. Although, the newer z-touch probing of these probes is great too!

closed 2026-09-03 by TitusLabs