protoloft/klipper_z_calibration #137: Perform Auto Z at KAMP Center

Dylan144GT, 2024-03-11:
Hi,

Is it possible to add a feature where the auto z is performed in the center of the [KAMP](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging) zone rather than in the middle of the bed. This is because it is good practice to print in different areas of the bed to ensure even wear, but the variance of the bed may be different in different KAMP zones.

Personally I find I need to adjust the z-offset whenever I don't print in the middle of the bed. One can argue that I don't have a flat bed, but I don't think that I am the only one in this position, especially on the more budget printers.

Thanks,

Dylan

TitusLabs, 2024-06-16:
Sorry, I don't use KAMP. But you can use the BED_POSITION parameter of the Z_CALIBRATION command. Would that be enough?

TitusLabs, 2026-09-03:
`CALIBRATE_Z BED_POSITION=x,y` does this — call it from your print start macro
with the centre of the KAMP area. Closing, since that covers the request.

closed 2026-09-03 by TitusLabs