protoloft/klipper_z_calibration #139: Move from clearance to safe_z_home when no [probe] section defined

DaStivi, 2024-03-22:
I've no [probe] section Defined as I use mentals/annex dockable probe.

`###################################################################
# 	Probe
#####################################################################

#[probe]
#pin: ~!PB7
#x_offset: 0
#y_offset: 25.0
#z_offset: 2
#speed: 20.0
#samples: 3
#samples_result: median
#sample_retract_dist: 3.0
#samples_tolerance: 0.006
#samples_tolerance_retries: 5

[dockable_probe]
pin: ~!PB7
dock_sense_pin: !PG15
x_offset: -4.3
y_offset: 21.0
#z_offset: 0
z_offset = 2.390   #texturierte oberfläche
#z_offset = 2.375   #glatte china PEI
#z_offset = 2.3050    #Schnitzelpei
samples: 3
samples_result: median
sample_retract_dist: 1.5
samples_tolerance: 0.006
samples_tolerance_retries: 5
dock_position: 54,352,15
#dock_angle: 270            #old-style radius approach - annexed_probe.py
#detach_angle: 180          #old-style radius approach - annexed_probe.py
#dock_safe_distance: 60     #old-style radius approach - annexed_probe.py
approach_position: 53,320
detach_position:  100,350
check_open_attach: true
dock_fixed_z: True
#safe_z_position: 232.3,350
speed: 3.0
travel_speed: 200
attach_speed: 35
lift_speed: 50.0
detach_speed: 50
dock_retries: 3
allow_delayed_detach: false
#pre_detach_gcode: G28 Z

#####################################################################
# 	Z Calibration
#####################################################################
[z_calibration]
# position on z-endstop for nozzle probing
#probe_nozzle_x: 235.6 #neue option seit august 2023, geändert 17.3.24 nach updates die alles kaputt gemacht gemacht to-be-deleted
#probe_nozzle_y: 350.0  #neue option seit august 2023, geändert 17.3.24 nach updates die alles kaputt gemacht gemacht to-be-deleted
nozzle_xy_position: 235.6,350.0
# position on z-endstop for switch probing
#probe_switch_x: 232   #neue option seit august 2023, geändert 17.3.24 nach updates die alles kaputt gemacht gemacht to-be-deleted
#probe_switch_y: 330.7   #neue option seit august 2023, geändert 17.3.24 nach updates die alles kaputt gemacht gemacht to-be-deleted
switch_xy_position: 232,330.5
# position on bed for print surface probing
# !!! this must be the relative reference point of the mesh, if using one !!!
# bed_xy_position: 175,175
#probe_bed_x: 175
#probe_bed_y: 175 #neue option seit august 2023, geändert 17.3.24 nach updates die alles kaputt gemacht gemacht to-be-deleted, wrude durch neue option zero_reference_position abgelöst
switch_offset: 0.385 # 0.670 ; smaller is more away from bed!
wiggle_xy_offsets: 0.5,0.5
offset_margins: -1.0,1.0
speed: 300
clearance: 10 #5
#samples: 5
#samples_tolerance: 0.01 #0.006
#samples_tolerance_retries: 10
#samples_result: median
#position_min: -5
#lift_speed: 5 #8
#probing_speed: 7 #10
probing_second_speed: 4
probing_retract_dist: 2
probing_first_fast: true
#end_gcode: detach_probe`

How should approach this if the option is removed?
BR steve

TitusLabs, 2024-06-16:
Uff, there is no probe at all? Not finding a probe alone throws an error. Missing parameters could be configured  explicitly. And for the positions are now three options each (https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It#command-calibrate_z)

DaStivi, 2024-08-10:
sadly i can't remember what i've changed back in March/April but i managed to fix it ... 🙈  i stumbled upon my own Post because of other changes that had been made to klipper on 9 April and 10 June... (eddy probe and general changes to probe )  that broke klipper again for me and i had unintenionally linked some issue here (#8) that had the same error message ("Internal error on command. "BASE_QUAD_GANRTY_LEVEL" -> DockableProbe instance has no attribute 'probing_move' )

closed 2024-08-10 by DaStivi