protoloft/klipper_z_calibration #140: Potential issues with oldham couplers

TmRxJD, 2024-04-12:
I've used this extension with great success for a long time now. Recently I upgraded a bunch of parts, entirely new toolhead and oldham couplers, and at first I wasn't getting any issues, was working consistently for speed benchy and getting perfect first layer every time. I tried changing filament and this is where I ran into issues and no matter what I do now the switch offset always returns negative. Going back to the old filament I could not get the same results I had before with the same values. I highly doubt it has anything to do with the filament I used but I checked absolutely every screw and made sure it was tight. The only thing I could think that would be causing issues is the allowed flex in the oldhams. I have the ones that slide where the leadscrews mount to the x gantry and springy motor couplers. I feel there may be some inconsistency however when I do the probe accuracy the tolerances are within 0.001mm. When making tiny movements trying to babystep the z I notice the pressure on the paper can vary slightly after returning to the same values, it seems like the flex is causing this. However it should still be good to get a working first layer. I have to get the nozzle touching the bed just slightly harder than I would like to even get a 0 switch offset to avoid the negative warning. The calibration macro returns a value I would like to see, about 0.1mm, and I have my endstop set to the suggested value, but the suggested switch offset is always 0 no matter what I do now and it prints way too close. Am I missing something here? changing the offset in settings makes no difference. I can't see where any of the inconsistency is coming from other than the couplers, but it shouldn't be enough from my measurements to cause me to not be able to print at all, curious if anyone else has reported similar issues and if it's just not possible to oldhams them with this extension. I don't think that's the case because originally it was working just fine. If it helps here's a video of my speed benchy from when it worked that includes my calibration process: https://www.youtube.com/watch?v=fWCt2gzLfiU if there's any other details about my setup or config that would be helpful please let me know.

TitusLabs, 2024-06-16:
Have you solved your problem in the meantime?

Unfortunately, I have no experience with leadscrews. But, a negative switch offset is not a problem at first. what counts is the position of the nozzle tip at z=0. If you don't like the negative offset, you need to adjust your configured z-endstop position.

TitusLabs, 2026-09-03:
Closing as stale. One correction to my 2024 answer: adjusting the Z endstop
position does not change the `switch_offset` the setup needs — all three samples
are toolhead coordinates and shift by the same amount, so only the reported
offset moves. The negative result points at the nozzle/switch-body measurement
instead:

https://github.com/protoloft/klipper_z_calibration/wiki/Troubleshooting#calculate_switch_offset-reports-a-negative-result

closed 2026-09-03 by TitusLabs