protoloft/klipper_z_calibration #144: AttributeError: 'PrinterProbe' object has no attribute 'sample_count'

Wizzkiss, 2024-06-10:
File "/home/pi/klipper/klippy/extras/z_calibration.py", line 110, in handle_connect
    self.samples = probe.sample_count
AttributeError: 'PrinterProbe' object has no attribute 'sample_count'

after update klipper to v0.12.0-229-g589bd64c

TitusLabs, 2024-06-10:
Please update to the recent version :+1:

closed 2024-06-11 by Wizzkiss

jojodroopy, 2024-06-11:
Hello, since this update my auto offset Z no longer works. Internal error is indicated

jojodroopy, 2024-06-11:
> Hello, since this update my auto offset Z no longer works. Internal error is indicated

after update klipper to v0.12.0-229-g589bd64c

jojodroopy, 2024-06-11:
> > Hello, since this update my auto offset Z no longer works. Internal error is indicated
> 
> after update klipper to v0.12.0-229-g589bd64c

[===== Config file =====.txt](https://github.com/user-attachments/files/15792230/Config.file.txt)

Wizzkiss, 2024-06-11:
Update z_calibration via moonraker Update Manager

TitusLabs, 2024-06-11:
Please update to the recent version :+1: 

> > > Hello, since this update my auto offset Z no longer works. Internal error is indicated
> > 
> > 
> > after update klipper to v0.12.0-229-g589bd64c
> 
> [===== Config file =====.txt](https://github.com/user-attachments/files/15792230/Config.file.txt)

Sorry, wrong place! That's not my z calibration plugin. Mine does not even work with a bltouch..

jojodroopy, 2024-06-15:
Ah... sorry, I'm bad at English and the translations are not always very reliable. For me the Bltouch works perfectly, it's just that since the last Klipper update, the auto Z offset no longer works.
So good luck with your machine... ;-)

ashinckley, 2024-06-18:
I've encountered this issue as well.  I'm trying to figure out how to fix it.
After updating Klipper (currently on v0.12.0-239-g433fcb6f)
I get the same error: "Internal error during connect: 'PrinterProbe' object has no attribute 'sample_count'"
I've updated your plugin via  update manager through moonraker to v1.1.0-0-gabbe9c4d

ashinckley, 2024-06-18:
Apologies.  I thing I fixed it.  Not sure what happened, but I simply reinstalled, and it seems to work now:
For those who might have also encountered this error, this seemed to do the trick.
cd ~
git clone https://github.com/protoloft/klipper_z_calibration.git
./klipper_z_calibration/install.sh

jojodroopy, 2024-06-18:
Ok, thank you very much I will try that

Good luck and nice day

jojodroopy, 2024-06-18:
Unfortunately, it doesn't work for me... I still have this internal error... I'll look again, but I don't know where. Thank you all the same for your help.

TitusLabs, 2024-06-18:
@jojodroopy I'm sorry, I meant to say that you were using a completely different plugin (auto offset z). Please contact the correct creator. My plugin has long since been fixed in this respect.

...ok, now you are using it. And that works for you? With a BLTouch??

jojodroopy, 2024-06-18:
No, that doesn't work. I use a Bltouch and before the Klipper update everything worked perfectly. Now even the input shaper also doesn't work anymore. What plugin do you use instead of auto offset z?

TitusLabs, 2024-06-18:
Your log tells me that you used this one: https://github.com/hawkeyexp/auto_offset_z

sanketss84, 2025-04-30:
> Update z_calibration via moonraker Update Manager

much thanks this solved my issue.