protoloft/klipper_z_calibration #146: After update no longer getting nozzle position from safe_Z_home

PhilBaz, 2024-06-12:
After update no longer getting nozzle position from safe_Z_home

Just updated today. When it was time to calibrate_Z it threw an error and asked me to either define a "nozzle_xy_position or use the NOZZLE_POSITION parameter". I use safe_z_home and have previously left nozzle_xy_position undefined and it has been taken the 'default' values from safe_z_home.

TitusLabs, 2024-06-12:
Yes, that was pretty stupid of me. I have now revised the complete position resolution. It has only been roughly tested so far. But you can test it in advance via the development branch...

PhilBaz, 2024-06-13:
@TitusLabs 

No problem. Ill wait for the release. Appreciate the fix, its nice to have it auto update with safe_z_home. Auto-Z is the best. Voron seriously wouldn't be as good without it. Thanks!

TitusLabs, 2024-06-16:
@PhilBaz Thank you very much for your kind words!

The fix is released :+1:

PhilBaz, 2024-06-16:
@TitusLabs 

Absolutely! Truly do appreciate your time and effort in development.

closed 2024-08-01 by TitusLabs