protoloft/klipper_z_calibration #147: Using possible z_endstop position value

Gil80, 2024-06-15:
Each time I'm using the HINT with the possible z_endstop_position value, I keep getting a larger possible z_endstop_position on the 2nd test and so on.. I keep chasing it.

1st test - I adjusted the value in my config and restarted the printer
```HINT: z position_endstop=1.845 - offset=-0.125644 --> possible z position_endstop=1.971
11:01 PM
Z-CALIBRATION: probe=6.736 - (switch=8.071 - nozzle=1.509 + switch_offset=0.300) --> new offset=-0.125644```

2nd test
```HINT: z position_endstop=1.971 - offset=-0.087360 --> possible z position_endstop=2.058
11:06 PM
Z-CALIBRATION: probe=6.761 - (switch=8.084 - nozzle=1.535 + switch_offset=0.300) --> new offset=-0.087360```

What am I doing wrong?

TitusLabs, 2024-06-16:
Where do you think you have a problem? You will never reach a offset calibration of 0.0 - otherwise, you would not need a calibration. It's just for safety to keep it small. And 0.087 is really small :+1:

Gil80, 2024-06-17:
Because the new offset keeps changing without any change I'm making.

```
HINT: z position_endstop=3.000 - offset=0.216039 --> possible z position_endstop=2.784
9:22 PM
Z-CALIBRATION: probe=6.759 - (switch=7.665 - nozzle=1.422 + switch_offset=0.300) --> new offset=0.216039
```
see the variations?
I'm trying to figure out why big changes in variations occur.

TitusLabs, 2024-06-17:
Hm, all three offsets are with a different z position endstop value. This is the absolute reference. Though the last one is way off compared to the offset. But still, z=0 should be correct...

Gil80, 2024-06-17:
Any ideas how to troubleshoot this?

TitusLabs, 2024-06-20:
To be honest, I don't even look at the resulting offset. This can change or fluctuate due to many circumstances. That's why I developed the plugin. As long as the result is correct with a correct first layer, everything is fine.

Otherwise, keep your z position endstop fix and observe everything else like temperature, plastic at the nozzle, switch to smooth PEI and so on...

TitusLabs, 2024-08-01:
With version 1.1.1, the hint for the position_endstop is only shown if the difference is greater 0.2mm

closed 2024-08-01 by TitusLabs