protoloft/klipper_z_calibration #148: Not seeing the zero_reference_position in bed_mesh section

Gray-Stone, 2024-06-23:
klipper_z_calibration commit: `abbe9c4dcd085913345d8b4551ca373769f1b108` 
klipper commit: `6d70050261ec3290f3c2e4015438e4910fd430d0` 

I have got this error: 

```
Cannot find a bed position! Either configure the bed_xy_position for z_calibration, the mesh's zero_reference_position, or use the NOZZLE_POSITION parameter.
```

Which upon looking into the code, I believe it triggered at the following lines: 

```
        if mesh is not None:
            if (hasattr(mesh.bmc, 'zero_ref_pos')
                and mesh.bmc.zero_ref_pos is not None):
                return mesh.bmc.zero_ref_pos
            elif (hasattr(mesh.bmc, 'relative_reference_index')
                  and mesh.bmc.relative_reference_index is not None):
                # TODO: remove: trying to read the deprecated rri
                rri = mesh.bmc.relative_reference_index
                return mesh.bmc.points[rri]
```

It seems to enter the first If statement, but didn't enter the internal if or elif sections. 

In my bed-mesh section I have used the the following config 
```
[bed_mesh]
speed: 300
horizontal_move_z: 11
##--------------------------------------------------------------------

##	Uncomment for 350mm build
mesh_min: 40, 40
mesh_max: 310,310
##--------------------------------------------------------------------
fade_start: 0.6
fade_end: 10.0
probe_count: 7,7 # Adaptive bed mesh should shrank this later
algorithm: bicubic
# adaptive: True
# relative_reference_index is removed. now we use a specific cord as reference.
zero_reference_position: 175.00, 175.00 
```

Work around: 
add `bed_xy_position` config to z_calibration section for now and the problem goes away.

confuzled-dev, 2024-07-04:
do you need to have a value with bed_xy_position?

gandalfb, 2024-07-21:
I set it to the same value as `zero_reference_position`

```
[bed_mesh]
speed: 300
horizontal_move_z: 10
##--------------------------------------------------------------------
##	Uncomment below for 250mm build
#mesh_min: 40, 40
#mesh_max: 210,210
#zero_reference_position: 125,125 #for use with stock z endstop

##	Uncomment for 300mm build
#mesh_min: 40, 40
#mesh_max: 260,260
#zero_reference_position: 150,150 #for use with stock z endstop

##	Uncomment for 350mm build
mesh_min: 40, 40
mesh_max: 310,310
zero_reference_position: 175,175 #for use with stock z endstop

##--------------------------------------------------------------------
fade_start: 0.6
fade_end: 10.0
probe_count: 9,9 # Values should be odd, so one point is directly at bed center
algorithm: bicubic

[z_calibration]
nozzle_xy_position: 232,350   #<X,Y position for clicking the nozzle on the z endstop - not needed if [safe_z_home] is used>
switch_xy_position: 232,325
#switch_xy_offsets:    #<X,Y offsets from the nozzle position for clicking the probe's switch body on the z endstop>
bed_xy_position: 175,175            #<X,Y position for probing the bed, for instance the center point - not needed if mesh with zero_reference_position is used>
switch_offset: 0.35           #<offset of the switch trigger (read the Switch Offset section!)>
start_gcode: ATTACH_PROBE
#before_switch_gcode: <macro name for attaching the probe AFTER probing the nozzle>
end_gcode: DOCK_PROBE
```

TitusLabs, 2024-08-01:
Finally, the fix is released :+1:

closed 2024-08-01 by TitusLabs