protoloft/klipper_z_calibration #152: Negative Switch Offset Troubleshooting

not-crazyramen, 2024-08-14:
Not an issue per se but what are some things to troubleshoot when I'm getting a negative switch offset? 

I have the probe and the endstop functioning properly and consistently. If I use either by themselves with a manual z offset I have no issues at all, but CALCULATE_SWITCH_OFFSET always gives me the negative offset error.

The probe body is firmly in place and the measured z height is consistent and repeatable. 

Forgive me if this is not the right place to ask for help with troubleshooting my setup, but I'm not sure where else to go.

SinisterRj, 2025-04-03:
I'm having the same problem. I really don't know why but my printer is "asking" for negative switch offset. I do the calibration and then measure the distance of the nozzle to the bed with a feller gauge and the difference is what I should compensate on the switch offset. I already tried to do the bed probing 5 mm in front of the end stop, to reduce any impact of Y/X axis twist and changing offset, but still no significative change

not-crazyramen, 2025-04-05:
Glad you linked kalico - I haven't heard of that fork before. 

It make sense that it shouldn't require/be possible to have a negative Z offset, so I wish there was more guidance on what could cause this. I'm not actuating the endstop pin on the actuator directly, so why are we getting a negative offset?

SinisterRj, 2025-04-07:
To be honest I don't know. I had two hypothesis: 1) any mechanical problem that is generating a twist on the toolhead, so the offset will change depending on the toolhead position. 2) another offset applied by klipper.
For the first one I tried to reduce the distance from the endstop probing the bed right in front of the switch, not in the middle of it, but with no clear change in my case. 
For the second one I tried to set as zero both endstop_position of endstop and z_offset of probe with no change also. So now I have no clue.

TitusLabs, 2026-09-03:
A negative result means the nozzle had to travel down further than the whole
current switch offset — more than the trigger travel of the switch, so the
difference is measured somewhere else. In order: the nozzle was not actually
touching the plate when you read the position; the probe's switch body doesn't
press the endstop plunger the same way the nozzle does; the probe doesn't seat
repeatably.

Note that adjusting the Z endstop position does not help: all three samples are
toolhead coordinates and shift together, so the reported offset moves but the
required `switch_offset` doesn't.

Full write-up:
https://github.com/protoloft/klipper_z_calibration/wiki/Troubleshooting#calculate_switch_offset-reports-a-negative-result

Closing — this is setup guidance, not a bug.

closed 2026-09-03 by TitusLabs