protoloft/klipper_z_calibration #158: Internal error on command:"CALIBRATE_Z" while using Bl-Touch

idontknowtom, 2024-10-06:
I have a modified Creality CR10S Pro with Bigtreetech SKR3 EZ and EBB36 using Klipper USB to CAN bus bridge and the EBB setup for CAN bus.

I use a Bl-Touch connected to the EBB36 and a Voron Sexbolt Z-Endstop on the front left corner of my bed.

I can home and BED_TILT_ADJUST using the Bl-Touch. But once I try to execute CALIBRATE_Z I get Internal error on command:"CALIBRATE_Z" and printer shutdown.

I have used this setup successfully in the past. Initially I left software outdated but once I got this error I figured it would be best to update everything. MCUs all have the latest Katapult and Klipper. 

Klippy log attached, most interesting is below:
[klippy.log](https://github.com/user-attachments/files/17268715/klippy.log)

> Internal error on command:"CALIBRATE_Z"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
Internal error on command:"CALIBRATE_Z"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/biqu/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/biqu/klipper/klippy/gcode.py", line 229, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/biqu/klipper/klippy/gcode.py", line 211, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 137, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/extras/z_calibration.py", line 152, in cmd_CALIBRATE_Z
    state.calibrate_z(switch_offset, nozzle_site, switch_site, bed_site)
  File "/home/biqu/klipper/klippy/extras/z_calibration.py", line 489, in calibrate_z
    switch_zero = self._probe_on_site(self.z_endstop,
  File "/home/biqu/klipper/klippy/extras/z_calibration.py", line 420, in _probe_on_site
    if self.probe.mcu_probe.query_endstop(time):
AttributeError: 'BLTouchProbe' object has no attribute 'mcu_probe'
MCU 'EBBCan' shutdown: Command request

idontknowtom, 2024-10-06:
Temporary fix is to rollback Klipper and Klipper Z Calibration to a point right before the probe.py changes.

Klipper ([49c0ad6](https://github.com/Klipper3d/klipper/commit/49c0ad6369670da574f550aa878ce9f6e1899e74)) - "motan: Fix logic error resulting in incorrect stepper phase graphing"

Klipper Z Calibration ([5489e7b](https://github.com/protoloft/klipper_z_calibration/commit/5489e7b7c56d392771056b6cbddb2db45fab98e2)) - "(error) messages revised"

Will have to look at code changes between the two to understand how to make it work with the Bl-Touch.
Might be more work than necessary as my printer currently is without enclosure but I'm planning to enclose it and move to a Cartographer.

TitusLabs, 2026-09-03:
BLTouch-style probes are not supported — the trigger point can't be measured
against the endstop, so the calculation has no valid basis. That won't change.

What does change in 2.0.0: the probe API is validated when Klipper connects, so
an unsupported probe fails at startup with a named error instead of raising an
`AttributeError` mid-move and shutting the MCU down. See #17 for the BLTouch
discussion.