protoloft/klipper_z_calibration #163: G92 instead of SET_GCODE_OFFSET ?

bombela, 2025-02-26:
Why not using G92 to update the Z homing after auto calibration instead of SET_GCODE_OFFSET ?

This way, SET_GCODE_OFFSET is still free to be used from a slicer, for when you want to tune the squish of the first layer. ABS requires to be quite squished into the bed, while PLA doesn't for example.

I am asking here before sending a pull request, there might be a reason for using SET_GCODE_OFFSET?

TitusLabs, 2025-02-27:
Please have a look here for the textured plate: https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Use-It You can still use SET_GCODE_OFFSET with Z_ADJUST in your PRINT_START macro for fine adjustments.

bombela, 2025-02-27:
Thank you. I have read this and should have mentioned it before. It might
be that my slicer is limited here. Please bear with me.

I don't think I can tune the PRINT_START parameters per filament in
PrusaSlicer.

I can add custom gcode to run on the first layer on filament activation (it
first layer, SET_GCODE_OFFSET Z_ADJUST=). But any filament change on the
first layer (like a color change) would apply another relative offset,
which is not good.

I personally don't plan to mix material (too slow printing swapping
filament without multiple tool heads). But I do have an occasional change
on the first layer for text printing. I would totally be fine with a single
relative offset application from the first filament used in the print.

But I haven't found a way to do that in PrusaSlicer.

Thank you for your time.

TitusLabs, 2025-02-27:
There are a lot of examples for the PRINT_START macro out there and how to configure it in PrusaSlicer. You can also create a variable which, for instance, gives the filament type - if this would be enough: in the Start GCode section you add this to the PRINT_START FILAMENT=[filament_type] And then, you can read it's value in the macro...

If you want to use SET_GCODE_OFFSET in the filament gcode, you need to set it back to Z_ADJUST=0 in the end/unload - if these values are only filament specific.

But, I have never heard about filament specific z height before. You do this with different textured sheet... but with filament - there are other parameters to tune like extrusion factor, first layer thickness or whatever.

bombela, 2025-02-27:
On smooth or sanded PEI (more matte finish), ABS needs to be pressed
against the bed to stick properly. In my testing, about 80μm closer to the
bed than PLA. And to be very clear, this is a comparison done with control
of all variables as much as possible (heat soak, air temp, bed top surface
temp, Z cal etc).

Now I agree this should be handled by the slicer directly. But PrusaSlicer
doesn't have any support for it. I do wonder though, how does the PrusaXL
and Core One deal with this then.

Changing the first layer height doesn't bring the nozzle closer to the bed.
Maybe what I really want, is a knob to adjust the extrusion multiplier of
the first layer depending of the material. PS doesn't have that out of the
box.

I will experiment with custom per filament gcode (through a klipper macro
for convenience of course) to increase the EM for a given filament. Maybe
this is all I need.

In all my testing on 150grit sanded PEI, I mostly found 3 squish levels:
- no squish: PETG, TPU (soft or hard)
- slight squish: PLA, Nylon, PC
- lotta squish: ABS, ASA, PE

In practice; for cosmetic appearances; I always want a slight squish, so
this leaves me with only two levels.

There is also a slight loss of dimensional accuracy by squishing more.
Which again makes me think maybe turning up the EM for the first layer
might be the answer.

In my experience with a textured plate, it is possible to have a single
squish setting but PLA will looks over squished in my experience. While on
a smooth PEI, PLA will grab the nozzle all over the place making an over
extrusion mess of the first layer.

In summary:

 - The more textured the plate. The closest to the bed you want to be to
press the plastic into the valleys. Watch out for when the tiny klicky
micro switch falls into a valley though ;)

 - The more waxy (like ABS that is viscous but not sticky), the more you
want to press it into the build plate for adhesion purposes.

Sorry for the long post. I am trying to get to the bottom of this problem.
People like me that print a lot of ABS are struggling with this pesky
squish difference. And I want one solution once and for all. I believe the
printer should have a perfect Z=0 when the nozzle is resting on the peak
surface of the bed surface. And the rest should be handled in the slicer.

bombela, 2025-03-01:
Here is the result of my testing the extrusion multiplier of the first layer vs lowering the Z closer to the bed.

### TL;DR

Adjusting first layer extrusion multiplier per filament material is similar to lowering Z to squish the first layer. But easier to setup, test and maintain.

### Test results

The machine was calibrated such that after auto z calibration, a gcode move Z=0 places the nozzle on the bed within 15µm. I used a 30µm thick piece of paper, and ran the sequence of homing, QGL and auto z many times, building up statistical confidence in the result. Following this, a mesh of 21x20 of the bed was recorded and used during printing. I have repeated this whole test at both room temp, and after 1h of heatsoak at 95°C top bed, 60°C air temp (actively controlled). I feel reasonably confident that I can get as close to a perfect Z=0 down to 15µm over any part of the bed.

Then I repeatedly printed a small stubby wall-like object in ABS. 100mm long, 2mm thick, 10mm tall. Close to the edge of the bed, encouraging shrinkage and warping with active cooling. I have also printed small flat squares for observing the effect of over extrusion on the first layer. A high EM might make a thin feature stick well on the bed, but make a mess of a large flat surface for example. All tests were made with elephant foot correction disabled. Leaving it to be tuned for cosmetic later.

The obvious: the closer to the bed, or the higher the extrusion multiplier, the higher the adhesion.

There is no difference in adhesion and cosmetic between lowering Z vs increasing EM. For example Z_ADJUST=-0.1 looks and sticks similar to FLEM=200% (first layer extrusion multiplier). Z_ADJUST=-0.05 looks and sticks like FLEM=125%. Etc. The only difference, is of course the height of the object. Lowering Z, lowers the height by the same amount. We are talking at most 100µm here, so in reality it doesn't matter that much, but if you want to be pedantic, FLEM is superior to lowering Z since it maintains dimensional accuracy. My printer seems accurate down to ±100µm on all dimensions. After adjusting for ABS shrinking ~0.4%. 

From my testing, I confirm that the higher the pressure against the bed, the better the adhesion. Note that I am using PrusaSlicer, which has provision to never shrink features on the first layer thinner than they could be printed when using elephant foot compensation. I propose for ABS to set the highest possible FLEM before the over extrusion becomes problematic and unsightly on bottom surfaces.

### PrusaSlicer Configuration

I believe that SuperSlicer has a FLEM setting. We can hope for the same in PS someday.
`M221` adjusts the extrusion multiplier on the printer.

#### Printer Start G-code

```gcode
M221 S100
```

#### Printer Before layer change G-code

```gcode
{if layer_num == 1}M221 S100{endif}
```

#### Filament Start G-code

Here the first layer extrusion multiplier is increased to 150%.

```gcode
{if layer_num < 1}M221 S150{endif}
```

#### Explanation

The printer EM is reset to 100% both the start of the print, and at the start of the second layer (first layer is 0, second layer is 1).
This leaves the filament start gcode free to override the EM for the first layer.

##### Filament change on the first layer

Notice that a filament change on the first layer will set the FLEM as expected as long as it is set in the filament start gcode. You could print ABS, PLA, PETG, and PE all on the same first layer if you really wanted. Every filament gets its own FLEM, not a fixed Z squish for all.

#### Advantages of the first layer extrusion multiplier

The printer is set to calibrate its Z precisely, once and for all. Z=0 means nozzle resting on the surface of the bed.. No need to adjust the auto z calibration SWITCH_OFFSET as a way to control pressure on the first layer. No need to fight auto z cal usage of SET_GCODE_OFFSET. No risk of combining random first layer height if printing different materials on the first layer.

#### Drawbacks

This config assumes full control of `M221`. The value is reset to 100% at the start of the print, and before the second layer. So live tuning is a bit annoying if that is something you need. It is possible to write some klipper macro to make this smarter. But we might as well patch our slicers to offer a first layer extrusion multiplier setting at this point.

### Conclusion

I still think a G92 should be used instead of SET_GCODE_OFFSET as a form of semantic correctness. But it doesn't matter for me anymore, since I am tweaking the EM of the first layer instead.

Thank you for reading thus far.

TitusLabs, 2026-09-03:
Keeping this open as the tracking issue for how the result is applied.

Partial answer available in 2.0.0: `offset_gcode` receives the calculated offset
as `params.Z` and takes over completely — when it's set, the plugin doesn't
touch `SET_GCODE_OFFSET` at all. So you can apply the offset your own way and
keep `GCODE_OFFSET` free for per-filament squish:

    [z_calibration]
    offset_gcode: MY_APPLY_Z_OFFSET Z={params.Z}

A built-in kinematic mode is what #175 proposes; the objections I raised there
apply. This issue stays the place to discuss it.