protoloft/klipper_z_calibration #164: Help: Only 1 nozzle probing immediately followed by running the `end_gcode`

spamwax, 2025-02-28:
I have set this great plugin in my other printers. For my new printer, using the following settings, the `CALIBRATE_Z` command moves the nozzle to top of nozzle, does only 1 probing, and immediately continues to run the `end_gcode` macro!

I have added some debugging messages to macros to see which one is being executed, and it shows that
- the `start_gcode` is run
- _one_ probing of the nozzle on the end-stop happens
- the `end_gcode` is executed

In the console, I just see one `CALIBRATE_Z: probe at 288.300,362.000 is z=1.103000` message, followed by my debug message from the `end_gcode`!
I am not sure how to debug this further. Nothing suspicious shows up in the logs or the console.

Here is my config:

```
[z_calibration]
nozzle_xy_position:   288.3, 362.0 # A X, Y coordinate (e.g. 100,100) of the nozzle, clicking on the z endstop.
switch_xy_position:   304.7, 353.3 # 306.7, 352.5 # A X, Y coordinate (e.g. 100,100) of the probe's switch body, clicking on the z endstop.
bed_xy_position:      177.5, 177.5 # <X,Y position for probing the bed, for instance the center point - not needed if mesh with zero_reference_position is used>
switch_offset:        1.856 # Note: Again, the smaller the switch-offset, the further the nozzle is away from the bed!
offset_margins:       -.3, 0.3
safe_z_height:        37.0 # The absolute z position in mm to move to before moving to the next position.
probing_speed:        0.5 # 2.5 # default from "stepper_z:homing_speed" section.
probing_second_speed: 0.5 # 1.0 # default from "stepper_z:second_homing_speed" section.
start_gcode:          _BEGIN_CALIBRATE_Z # <macro name for attaching the probe>
before_switch_gcode:  PREPARE_FOR_PROBING_KAICLID # <macro name for attaching the probe AFTER probing the nozzle>
end_gcode:            _END_CALIBRATE_Z # <macro name for docking the probe>
speed:                120
samples:              2
```

The gcode macros are pretty simple:

```
[gcode_macro _BEGIN_CALIBRATE_Z]
gcode:
    ASSERT_PROBE_STOWED
    M118 🍄 STARTED z_calibration 🍄

[gcode_macro _END_CALIBRATE_Z]
gcode:
     M118 Started 💀💀 _END_CALIBRATE_Z 💀💀
     STOW_PROBE

[gcode_macro PREPARE_FOR_PROBING_KAICLID]
gcode:
    M118 Started 🍄 PREPARE_FOR_PROBING_KAICLID 🍄
    DEPLOY_PROBE
```

The only difference, between this printer and others is that the z-endstop uses a pressure sensor instead of of mechanical switch. It is working just fine as I can normally home the printer, and do a multi-sample Z-Homing on it without any problem.

Nilleri, 2025-02-28:
I have a very similar problem. It will attach the klicky probe and probe 5 times using the nozzle on the endstop, and then dock the probe again. I really hope we can get some clarification for this.

spamwax, 2025-02-28:
Interesting! What is your printer?
The thing for me is that I have to undock the probe to safely touch the nozzle on endstop, so it is not working even when there is no probe involved.
I was doing a little bit of debugging and it seems the function call [`curpos = phoming.probing_move(mcu_endstop, pos, speed)`](https://github.com/protoloft/klipper_z_calibration/blob/9f16ebc010e117ae81f9402fba90ed8c523726ae/z_calibration.py#L329C13-L329C67) is being abruptly abandoned!

spamwax, 2025-03-03:
It seems this is related to timing issue of the pressure sensor response time (going from actuated to de-actuated) and the next homing move that, somehow, happens too quickly before the pressure sensor has fully gone down to LOW voltage. 
I am suspecting some sort of syncing needs to happen between MCU code that moves the steppers and the code that reports status of endstop

TitusLabs, 2026-09-03:
Two things here.

Your own diagnosis is most likely right: the pressure-sensor endstop hasn't
gone low again by the time the next homing move starts. That's a Klipper timing
issue, not something the plugin can synchronise.

But the missing error message is on me: in the current release `end_gcode` runs
in a `finally` block, so if it fails it replaces the original exception — which
looks exactly like "one probe, then end_gcode, no error". 2.0.0 keeps the
original error and only logs an `end_gcode` failure. Please retest on 2.0.0;
you should get a real message then.

closed 2026-09-03 by TitusLabs