protoloft/klipper_z_calibration #165: Useful manual

pasha4ur, 2025-04-14:
Hello

Could you write a good useful manual about configuring and using your software?

In [this manual](https://github.com/protoloft/klipper_z_calibration/wiki/How-To-Configure-It#configurations) you didn't write even about the file's name where we should add this code. 

Please, give us at least some working examples for some stock printers like Voron Trident, 2.4, etc.

3DCoded, 2025-06-05:
@pasha4ur ,

You put the `[calibrate_z]` section in your `printer.cfg` or a file included by it.

TitusLabs, 2026-09-03:
Fair point. The section goes into `printer.cfg` or any file included by it, as
@3DCoded said. I'm reworking the Wiki alongside the 2.0.0 release and will make
that explicit, plus a complete working example.

Closing here — the Wiki is the place for it.

closed 2026-09-03 by TitusLabs