protoloft/klipper_z_calibration #166: g28 not grabbing the probe

mbohben, 2025-04-14:
my klicky was running flawlessly without [z_calibration] enabled, seems like my homing overide didnt work well with that.
just like the tittle above, i've changed the endstop pin to the segsbolt endstop too since probe:virtual endstop wont work with [z_calibration] enabled.

there are my [z_calibration] config:

```
[z_calibration]
nozzle_xy_position: 142,241 #default from home_xy_position of safe_z_home
#   A X, Y coordinate (e.g. 100,100) of the nozzle, clicking on the z endstop.
switch_xy_position: 135,205 #optional when switch_xy_offsets is used
#   A X, Y coordinate (e.g. 100,100) of the probe's switch body, clicking on
#   the z endstop.
bed_xy_position: 115,115 #default from zero_reference_position (or relative_reference_index) of bed_mesh
#   A X, Y coordinate (e.g. 100,100) where the print surface (e.g. the center
#   point) is probed. These coordinates are adjusted by the
#   probe's X and Y offsets. The default is the zero_reference_position which
#   replaces the deprecated relative_reference_index
#   of the configured bed_mesh, if configured. It's possible to change the zero
#   reference position at runtime or use the GCode argument BED_POSITION of CALIBRATE_Z.
wiggle_xy_offsets: 0,0
#   After probing the nozzle and retracting, move x some distance away and
#   back. Useful to prevent the z endstop pin sticking to the nozzle and
#   being pulled out of the assembly. Can be negative. Defaults to zero to
#   disable it. Can be combined in x and y to move diagonally. Be careful
#   to not move your nozzle out of range!
switch_offset: 0.5
#   The trigger point offset of the used mag-probe switch.
#   A larger value will position the nozzle closer to the bed.
#   This must be determined manually. More on this later
#   in this section..
offset_margins: -1.0,1.0
#   The minimum and maximum margins allowed for the calculated offset.
#   If the offset is outside these values, it will stop!
#   The margin can be defined as "min,max" e.g. "-0.5,1.5" or by just one
#   value e.g. "1.0" which translates to "-1.0,1.0" (which is also the default).
safe_z_height: 10 #default is 2 * z_offset from the "probe:z_offset" section
#   The absolute z position in mm to move to before moving to the next
#   position. The default is two times the z_offset from the probe's
#   configuration. The minimum safe z height is 3mm.
position_min: 10 #default from "stepper_z:position_min" section.
#   Minimum valid distance (in mm) used for probing move. The
#   default is from the Z rail configuration.
speed: 50
#   The moving speed in X and Y. The default is 50 mm/s.
start_gcode:
    ATTACH_PROBE
before_switch_gcode:
    G90
    G1 Z10 F1200              # Lift nozzle before moving
    ATTACH_PROBE              # Ensure Klicky is attached
end_gcode:
    G90
    G1 Z10 F1200              # Lift nozzle after calibration
    DOCK_PROBE                # Dock Klicky after use
```

TitusLabs, 2026-09-03:
Homing itself is not driven by this plugin — the docking is done by your own
homing override / Klicky macros, and that's where the probe fails to attach.
`[z_calibration]` only calls `start_gcode`, `before_switch_gcode` and
`end_gcode`.

Closing as stale; please reopen with your homing macros and a `klippy.log` if
you're still stuck.

closed 2026-09-03 by TitusLabs

mbohben, 2026-09-07:
> Homing itself is not driven by this plugin — the docking is done by your own homing override / Klicky macros, and that's where the probe fails to attach. `[z_calibration]` only calls `start_gcode`, [...]

its solved, thanks in advance